beekeeb/piantor #1: Only USB-C ports

Lumi571, 2022-11-04:
Would it be possible to create a version of the Piantor whose TRRS ports are replaced by USB-C ports?

l4u, 2022-11-04:
Thank you for the suggestion, Lùmi! To keep this beginner-friendly for hand soldering, the TRRS ports will not be replaced by USB Type-C ports. It might happen if there will be a PCBA version. You might be interested in https://github.com/KarlK90/yaemk-split-kb

Lumi571, 2022-11-04:
Thanks for taking time to respond to me. :)

It seems to me that I have read that complicated soldering, such as that of the USB-C ports, can be carried out by machines in the factory if a PCBA file is provided. In this case, why not use it to take advantage of the qualities of a USB-C port while saving yourself the task of soldering it?

PS: The Yaemk-split-kb unfortunately lacks shock spacing and strong column stagger.

l4u, 2022-11-05:
You can apply the schematic from yaemk-split-kb to this project to build your own version.

closed 2022-11-05 by l4u