beekeeb/piantor #12: Disconnecting the TPRS cable when the keyboard is plugged in fries the chip

SadraMoh, 2023-07-30:
Unfortunately I found this out the hard way when I was building my own keyboard using the DIY kit from beekeeb.

Now I know there are already warnings in the beekeeb docs warning against this, but it is inevitable that at some point this will happen by mistake. I acknowledge that it is mostly my fault this has happened and have no intention to blame the designers.

I would rather ask, is there a way to make this safer so this doesn't happen again to other people?

viyoriya, 2023-08-01:
If you design PCB place the routing of vcc/3v3 as last pin in TRRS then this wont happen

philong, 2023-08-15:
I've read that TRRS tip should be VCC and last ring should be ground. That way, you don't hit current until fully inserted, and shorting ground to VCC should be nearly impossible.

I wonder whether the Cantor is also affected.

jojo599k, 2023-08-19:
The cantor has following configuration:

- TIP (2) GND
- RING1 (3) RX
- RING2 (4) TX
- SLEEVE (1) VCC

The left and right piantor have the following configuration:

- TIP (2) VCC
- RING1 (3) GND
- RING2 (4) TX
- SLEEVE (1) RX

According to this comment on reddit https://www.reddit.com/r/MechanicalKeyboards/comments/b1s7c9/comment/einz0xw the configuration should be the following:

- TIP (2) VCC
- RING1 (3) TX / RX
- RING2 (4) RX / TX
- SLEEVE (1) GND

which is the reverse for GND and VCC for cantor configuration. That is also the configuration @philong is recommending.

I'm not quite sure what @viyoriya configuration would look like.

According to ChatGPT, the default configuration for transmitting data with a TRRS connector would be:

- TIP (2): RX (Receive Data)
- RING1 (3): TX (Transmit Data)
- RING2 (4): GND (Ground)
- SLEEVE (1): VCC (Voltage/Power Supply)

Which is the reverse of the piantor configuration.

I have sadly no experience with PCB design and wiring, so i cant give the right answer here. It would be nice if someone could clarify this. And of course if someone could update the pcb and create a PR.

viyoriya, 2023-08-22:
@jojo599k I used 

    TIP (2) GND
    RING1 (3) RX
    RING2 (4) TX
    SLEEVE (1) VCC

same as cantor attached the pcb and sch diagram
 [](url)
[image]

hxse, 2023-08-31:
Really important hope that this problem can be solved.

pcky, 2024-02-14:
As a workaround i integrated an USB-C Socket as replacement into the 3D-printed case of my Piantor keyboard:
https://www.printables.com/model/764951

renanqts, 2024-04-15:
@pcky could you please let me know how you soldered this usb socket behind it?