beekeeb/piantor #15: question about flash the borad, which flash file?

sIHURs, 2023-08-21:
I would like to ask a question about the file "beekeeb_piantor_rpi_pico_vial.uf2." Should this file be loaded into both the left and right microcontrollers (raspberry pi pico)? After loading, my right-hand device actually has the layout of the left hand?

sIHURs, 2023-08-21:
and only the keyboard connected to the USB side is working properly, while the other half shows no response when plugged in.

AlexeyKhrenov, 2023-10-06:
I guess, you already figure out the answer :) anyway, for newcomers: 
you need to put .uf2 on 2 parts separately while they're not connected with a cable. Now you can join them together and to PC. The side connected to the PC becomes master.