beekeeb/piantor #21: Piantor stopped working

VinciTec, 2023-12-18:
Anyone have an idea what is wrong?

I have happily used a Piantor keyboard for a couple of months now, and regularly take it with me to other places in a protective bag from Beekeeb. Now I plugged it in again and is seems to have stopped working correctly.

First of all, I'm aware unplugging the TPRS cable whilst powered can cause components on the microcontrollers to be damaged, so I have been very careful each time when packing away the keyboard. As far as I am aware I have never unplugged the TPRS cable before unplugging the USB C cable. However, as there seems to be no other change from the last time the Piantor still worked, maybe this happened without me realizing. Although, I don't believe I did.

Anyway, when connecting power to the (normal left-side of the) keyboard now the power LEDs glow much weaker than before, and fluctuate slightly. Connected this way, my computer receives no signals form the keyboard when keys are pressed. However, very infrequently a couple of key presses go through (the LED seems to also glow brighter when this happens).
The weird thing, to me at least, it that when I connect the USB C to the right side of the Piantor, both LEDs glow as normal and all key presses register fine. However, when plugged in on this side the key mapping is all wrong as the QMK config of the Piantor is configured relative to the left side.
Moreover, when just using the left half of the Piantor (meaning I connect the USB C to the left side without using the TPRS cable to connect to the right side) the LED glows at normal brightness and all key presses register correctly. But in this setup I am not able to use the right side, so I cannot use the keyboard in this way.
By the way, I can still upload new firmware without any problems.

It seems I could use the keyboard by connect to the right side and reconfiguring the keymapping. However, clearly something is wrong, and I do not want to damage the micro controller further.

Does anyone have an idea what is wrong with my Piantor? What could have caused it? Might it be a firmware problem that can easily be resolved? Or can I replace a component?
Thanks for any help.

VinciTec, 2023-12-28:
I imagine the problem must lie with (one of or both) the microcontrollers. Is there any way I can troubleshoot this? Test which one is broken so I can possibly replace it?

quirq, 2024-02-07:
I've had the same issue a number of months after I got my keyboard - but I received the v2 with USB-C on both halves.

I have tried to reflash the firmware on the RP2040, but it loses it after unplugging the keyboard.

Expect that the issue is with the RP2040 or the board in general, but haven't figured it out yet.