beekeeb/piantor #22: One side of the keyboard not working

Hentuloo, 2024-03-06:
Hi, I encountered a problem where only one part of the keyboard was working after assembly, depending on the connected power source, as shown in the picture where the LED only lights up on one side: 

[image]

I'm using Tzt raspberry pi pico board rp2040 16MB (dual-core 264kb arm low-power mikro computer hoch leistungs Cortex-M0 + prozessor) which was purchased on AliExpress.

While searching for a solution, I came across a diagram:

[image]

It appears that the Vin and Vout pins are connected/shorted, whereas on my boards they were not connected. After soldering these pins on both parts, the keyboard started functioning 🎉

[image]

The problem lies in the fact that the second board had insufficient voltage, which prevented it from starting up.

[image]

I have created this issue requesting to add some information about it on the BeeKeeb homepage. It may help people like me who have bought similar microcontrollers.

magicalpuffin, 2024-03-07:
This is a really great find. I bought a Aitexm Raspberry Pi Pico Board RP2040 TYPE-C 16 MB on Aliexpress and I was having similar issues where I could only get one half connected. 

I tried soldering the Vin and Vout pins and was able to get both LEDs to light up, however, I couldn't get the Vial firmware to work.

I already permanently damaged one microcontroller when I tried to remove it so I'll have to abandon my build but this does have potential to help others.

[image]

Hentuloo, 2024-03-07:
I also had other issues with the keypads that I didn't mention: 

1. one of the keypads refuses to connect to the computer at all; it shows a message that the device is unknown. I tried various things, but it's likely that it needs to be replaced (I recommend ordering a spare from AliExpress right away).

2.  I also encountered problems with the remaining keypads. Initially, I used the beekeeb_piantor_rpi_pico_vial.uf2 software available on the beekeeb website (https://docs.beekeeb.com/piantor-keyboard#firmware-for-clearing-the-flash)

[image]

but the keyboard didn't appear in the Vial software. In the web version (https://vial.rocks/) I noticed the console error "device read timed out!" (screen)

[image]

After several attempts, I decided to install the beekeeb_piantor_vial.uf2 software, which is the third option from the top, and both Vial web and desktop versions **started working**. 

[image]

Before flashing any software, I cleared the memory using flash_clear.uf2 "Firmware for clearing the flash," which is also available on beekeeb (https://docs.beekeeb.com/piantor-keyboard#firmware-for-clearing-the-flash).

mumblingdrunkard, 2024-03-21:
Just adding this for anyone searching for "YD-RP2040 Piantor".
My friend bought YD-RP2040 controllers and this same "hack" works on this.
Shorting Vin and Vout on the master side lets the other side receive power.

This happens because the original Pico has Vbus and Vsys instead of Vout and Vin, respectively.  Vsys can source current and the Piantor PCB has a trace running from Vsys to the TRRS port and uses that to power the non-master side.
On the YD-RP2040 (and whatever the controllers here are) this pin is replaced with Vin, which cannot source current, and delivers no power to the TRRS port, although it can receive power from it.

This is/should be safe as you're only bypassing the diode that goes from Vin to Vout, but experiment with your cheap knockoff microcontrollers at your own risk.
I _believe_ it's safe to short both sides provided you only power one side at a time, but you only need the short on the master side.