beekeeb/piantor #27: Right side not working (sometimes and partially)

sollted, 2025-03-05:
I love the keyboard but sometimes right half (slave) stops working, the led is still on but input doesn't work for some time from seconds to minutes it can even stop working completely. 
When this happens i disconnect keyboard and reconnect or press the reset button but mostly waiting is enough.
It is extremely irritating. I switched from bluetooth corn keyboard exactly because issues like this one.

For replicating:

- flashing .uf2 to left half (disconnecting then holding BOOTSFL while connecting.
- I am getting firmware from QMK configurator. in safari but tried chrome (it should not have any effect)
- Mac m1 pro this issue persisted throughout all versions of OS
- Czech keyboard layout (if it matters)

Thanks for response in advance.

mertoldi, 2025-05-10:
I'm encountering exactly the same problem, this has only started happening recently, nothing has changed on the computer.  The keyboard is used on a windows 11 machine.

sollted, 2025-05-15:
I made the problem more manageable by increasing timeout in QMK configuration. (this didn't fix it but made it occur less often). But I resoldered the keyboard (for 3rd time) and now it works.