beekeeb/piantor #28: Piantor Right Half (slave) Not Working

Heinrich-XIAO, 2025-04-27:
It was working fine until it randomly stopped working. I feel like it has to be a hardware issue since software is generally predictable. I notice that when the issue was first happening, it would register one or two keys at the start and then stop. It is no longer doing so, either because I haven't done thing that would replicate it or it truly stopped. I've checked the TRRS cable connecting them and there are no shorts and everything connects correctly within the cable. After that, I checked if the pins that are supposed to connect for the TRRS cable were connected, and indeed they were. The left side is a WeAct that came with the keyboard and the right side (slave and not working) is a RPI Pico (not 2). Plugging in the right side on its own works fine except the keys are different which is expected.

Firmware: QMK then tried Vial and it still didn't work.

What should I try next?

mertoldi, 2025-05-10:
I have the same problem , on windows 11 with 2 raspberry pi pico's

nsmorales, 2026-09-18:
I recently faced the same issue. In my case the right side controller was working by itself but not when it was supposed to be the slave or vice versa. The right rpi had suffered some kind of damage were the pins used for communication between it and the other half were simply broken. I had to get another RPI did a dirty wiring directly to the master and confirmed that it was working as a slave. So ended then desoldering the partially broken rpi with the new one and everything works now.